inivation/libcaer #1: Windows support

llongi, 2016-05-10:
Windows is not supported currently.
It might be in the future, depending on what exactly VS2016 all supports of the C11 standard. Of particular concern is stdatomic.h, though that could be implemented using InterlockedExchange() if needed. c11threads.h would need a Win32Threads implementation like it's done now for Pthreads, but that's easy.
I'll keep this bug open and update it when there are any news concerning this.

llongi, 2016-08-26:
Annnd we have it. :-) Using MinGW and tested there, build and use seems to work.
Using the DLLs/headers in Visual Studio should work too, but hasn't yet been tested.
If you experience issues, please open a new bug with details, thanks!
And a big thank you to Greg Cohen, for his help in figuring out and testing the Windows build!

closed 2016-08-26 by llongi

Thanh-Binh, 2016-08-27:
@llongi 
I have just download the current version and tried to build with VS2015 + cMake 3.6.1. Here is error:

###############
The C compiler identification is MSVC 19.0.24213.1
The CXX compiler identification is MSVC 19.0.24213.1
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
Check for working C compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
Detecting C compiler ABI info
Detecting C compiler ABI info - done
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe
Check for working CXX compiler: C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC/bin/x86_amd64/cl.exe -- works
Detecting CXX compiler ABI info
Detecting CXX compiler ABI info - done
Detecting CXX compile features
Detecting CXX compile features - done
fatal: Not a git repository (or any of the parent directories): .git

CMake Warning (dev) at CMakeLists.txt:34 (IF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:36 (ELSEIF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:38 (ELSEIF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at CMakeLists.txt:40 (ELSEIF):
  Policy CMP0054 is not set: Only interpret if() arguments as variables or
  keywords when unquoted.  Run "cmake --help-policy CMP0054" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.

  Quoted variables like "MSVC" will no longer be dereferenced when the policy
  is set to NEW.  Since the policy is not set the OLD behavior will be used.
This warning is for project developers.  Use -Wno-dev to suppress it.

Check if the system is big endian
Searching 16 bit integer
Looking for sys/types.h
Looking for sys/types.h - found
Looking for stdint.h
Looking for stdint.h - found
Looking for stddef.h
Looking for stddef.h - found
Check size of unsigned short
Check size of unsigned short - done
Using unsigned short
Check if the system is big endian - little endian
Check size of size_t
Check size of size_t - done
Check size of void *
Check size of void * - done
Looking for pthread.h
Looking for pthread.h - not found
Found Threads: TRUE  
Could NOT find PkgConfig (missing:  PKG_CONFIG_EXECUTABLE) 
CMake Error at C:/Program Files/CMake/share/cmake-3.6/Modules/FindPkgConfig.cmake:479 (message):
  pkg-config tool not found
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.6/Modules/FindPkgConfig.cmake:597 (_pkg_check_modules_internal)
  CMakeLists.txt:152 (PKG_CHECK_MODULES)

Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
Project version is: 2.0.0-r
Compiler is Clang: FALSE
Compiler is GCC: FALSE
Compiler is IntelCC: FALSE
Compiler is MS VisualC: TRUE
OS is Unix: FALSE
OS is Linux: FALSE
OS is MacOS X: FALSE
OS is Windows: TRUE
System is big-endian: 0
Thread support is PThreads: FALSE
Thread support is Win32 Threads: TRUE
C flags are:  /DWIN32 /D_WINDOWS /W3
CXX flags are:  /DWIN32 /D_WINDOWS /W3 /GR /EHsc
Include directories are: C:/Projects/NeuromorphicSensors/libcaer/include/
Library directories are: 
Linked libraries are: 
Compiled source files are: ringbuffer/ringbuffer.c;log.c;events.c;frame_utils.c;device.c;dvs128.c;davis_common.c;davis_fx2.c;davis_fx3.c
Final install libdir is: C:/Program Files/libcaer/lib
Final install includedir is: C:/Program Files/libcaer/include/libcaer
Configuring incomplete, errors occurred!
See also "c:/Projects/NeuromorphicSensors/libcaer/VS2015/CMakeFiles/CMakeOutput.log".
See also "c:/Projects/NeuromorphicSensors/libcaer/VS2015/CMakeFiles/CMakeError.log".

llongi, 2016-08-27:
Building with Visual Studio is not supported. We only support MinGW64, please take a look at the detailed instructions in README.Windows. The MSVC compiler lacks several basic features that we use, supporting it would require us to write lots of emulation code, which the MinGW people already did, for stdatomic.h, for pthreads, for VLAs, ..., either in their compiler or their libraries.
Using the resulting DLLs with Visual Studio on the other hand should work, but we haven't tested that yet.