inivation/libcaer #5: make install failed

Thanh-Binh, 2016-10-05:
by compiling the current version I have an error like
cmake -DCMAKE_INSTALL_PREFIX=/usr .
make install

[100%] Built target caer
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/lib/x86_64-linux-gnu/pkgconfig/libcaer.pc
CMake Error at cmake_install.cmake:36 (file):
  file INSTALL cannot copy file "/home/binh/Neuromorphics/libcaer/libcaer.pc"
  to "/usr/lib/x86_64-linux-gnu/pkgconfig/libcaer.pc".

Makefile:61: recipe for target 'install' failed
make: *** [install] Error 1

Could you help me?

llongi, 2016-10-05:
Are you running as root? To copy files to system directories you'll need to run make install as root or via sudo.

Thanh-Binh, 2016-10-05:
you are right. I just follow your README without thinking.
Please change your README
make install
into
sudo make install
best thanks

closed 2016-10-05 by llongi