inivation/libcaer #6: [enh] Packet constructor using data buffer from socket

breznak, 2016-10-10:
Proposal: Would it be possible to add an EventPacket constructor using some existing data? 

https://github.com/inilabs/libcaer/blob/master/include/events/polarity.h#L87 

I'm looking at networking/Sockets and the `tcpststat` util, where: 
```
              if (eventValid > 0) {
                        void *firstEvent = caerGenericEventGetEvent(header, 0);
                        void *lastEvent = caerGenericEventGetEvent(header, eventValid - 1);

                        int32_t firstTS = caerGenericEventGetTimestamp(firstEvent, header);
                        int32_t lastTS = caerGenericEventGetTimestamp(lastEvent, header);

                        int32_t tsDifference = lastTS - firstTS;

                        printf("Time difference in packet: %" PRIi32 " (first = %" PRIi32 ", last = %" PRIi32 ").\n", tsDifference,
                                firstTS, lastTS);
                }

```

I think a nicer approach would be to construct a Packet, then use the `LOOP_VALID_EVENTS...` and being able to use the `caer*EventGetXXX(evt)`. 

What do you think?

breznak, 2016-10-18:
Closing as I figured a clean way: 
- converted the buffer to `PolarityPacket`
- used `caerPolarityPacketGetEvent()`

closed 2016-10-18 by breznak

llongi, 2016-10-18:
Yes, that's the idea. You get a block of data, a packet. You can cast it to a caerEventPacketHeader and then use the GetType() header function to find out what exactly it is (and if needed GetEventSize() and GetEventNumber() to guess how big it is and if you need to fetch any more bytes, tcpststat.c does this for example). Then you just cast the whole thing to the type you found out, and use the usual functions for that event type. The caerGenericEvent*() functions are there only for some very specific, generic use-cases: getting validity and timestamp without looking at the type first, which is used in some very specific input/output/ordering cases, but never really when processing and working with data: at that point you usually know what you're working on.