inivation/libcaer #7: Make docs browsable online

breznak, 2016-10-10:
Many projects offer docs online as github pages, this would be very convenient. Eg. https://github.com/zeromq/zmqpp

llongi, 2016-10-17:
Done. Docs now available at http://inilabs.github.io/libcaer/ also updated PDF at http://inilabs.github.io/libcaer/libcaer_api_manual.pdf

closed 2016-10-17 by llongi

breznak, 2016-10-17:
..is this hooked to update with new PR to libcaer?

llongi, 2016-10-18:
No, I'll rebuild them manually as needed for now. libcaer is widely stable. I also don't necessarily want it to always reflect the latest unstable changes until we've figured them out.

breznak, 2016-10-18:
Ok. 
Reopening as `make doc/pdf` from README does not work..

llongi, 2016-10-18:
It does here, that's how I generated those docs. Just tried it again, works fine.
Though make doc/pdf support is optional and only works if Doxygen is installed on your system.
Run cmake and check the output there, if it finds it.

breznak, 2016-10-18:
Sorry, my doxygen is indeed missing. That resolves it. thanks! 
`-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE)`