inivation/libcaer #8: [Bug] Generic Event includes [+solution]

breznak, 2016-10-14:
I run into the following error when compiling C++ code with `libcaer` C includes: 

```
g++ -std=c++11 -o test NetworkReader.cpp -I. -Lpthread
/tmp/ccnLIAZm.o: In function `caerGenericEventGetEvent':
NetworkReader.cpp:(.text+0x21b): undefined reference to `caerLog'
/tmp/ccnLIAZm.o: In function `std::thread::thread<NetworkReader::start()::{lambda()#1}>(NetworkReader::start()::{lambda()#1}&&)':
NetworkReader.cpp:(.text._ZNSt6threadC2IZN13NetworkReader5startEvEUlvE_JEEEOT_DpOT0_[_ZNSt6threadC5IZN13NetworkReader5startEvEUlvE_JEEEOT_DpOT0_]+0x7d): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
``

I found the following errors: 
- missing include in `common.h`: https://github.com/inilabs/libcaer/blob/master/include/events/common.h#L18 does not exist. 
- the linker complains about `caerLog` (from `log.h`) which is not included in the common.h

breznak, 2016-10-17:
> https://github.com/inilabs/libcaer/blob/master/include/events/common.h#L18 does not exist.

the file does not exist in the repo, but the installed location contains `libcaer.h`, is it created during the installation? 

Also, the err occurs only when linking C++ code, not when compiling pure C

llongi, 2016-10-17:
It is created at build time. All files are laid out so that this works. libcaer.h includes log.h, and libcaer.h is included as first thing in common.h, which is included by the other events. So it's all correct. All includes are relative to allow for easier development, so that it works even without having to do a full make install to /usr/include. I've never seen problems with any of this.
What do you include in NetworkReader.cpp and where is it located?

breznak, 2016-10-17:
Thank you, so looks like I have 2 problems: 

* minor: I think the `libcaer` should be used only by including `libcaer.h`, right? All other are included from it (eg. `polarity.h`). But that does not seem to be the case for me: 

```
#include <libcaer/libcaer.h>
...
./caer_client.c:135:51: error: use of undeclared identifier 'CAER_EVENT_PACKET_HEADER_SIZE'
                if (!recvUntilDone(listenTCPSocket, dataBuffer, CAER_EVENT_PACKET_HEADER_SIZE)) {
```

* conflicting `string`, `cstring`, the error in `caerLog` seems to come from this conflict. Clang++-3.8 is more verbose and helpful, I get a bunch of these: 

```
$ clang++-3.8 -std=c++11 -o test NetworkReader.cpp -lpthread 
In file included from NetworkReader.cpp:2:
In file included from ./NetworkReader.hpp:12:
In file included from ./caer_client.c:20:
In file included from /usr/local/include/libcaer/libcaer.h:20:
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/cstring:100:3: error: conflicting types for 'memchr'
  memchr(void* __s, int __c, size_t __n)
  ^
/usr/include/string.h:92:14: note: previous declaration is here
extern void *memchr (const void *__s, int __c, size_t __n)
             ^
```

breznak, 2016-10-17:
Project structure: 

Aim (TCP client for C++), rip-off of `utils/tcptstat` :) 

* NetworkReader.cpp: 
```
// reads Socket data on TPC, stores in queue and allows to pop
#include "NetworkReader.hpp"

//! namespace caer { // cannot use namespace due to C
using namespace std;
```

* NetworkReader.hpp:

```
#pragma once

#include<thread>
#include "ProducerConsumerQueue.h"

//#define caerLog(x) //dummy fix

#ifdef __cplusplus
extern "C" {
#endif
 #include "caer_client.h"
 #include "caer_client.c" //MUST include the .c file, NOT .h, otherwise linking fails
#ifdef __cplusplus
}
#endif
```

 * ProducerConsumerQueue.hpp (FB's folly lib)

```
#pragma once

#include <atomic>
#include <cassert>
#include <cstdlib>
#include <memory>
#include <stdexcept>
#include <type_traits>
#include <utility>

namespace folly {
```

* caer_client.c (aka. tctststat)

```
/*
 * C Socket client (TCP) for cAER, modified from tcptsstats util
 * tcpststat.c
 *
 *  Created on: Jan 19, 2014
 *      Author: llongi
 */

#include <stdlib.h>
#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
//#include <string.h> //FIXME commented out, still errs (?!)
#include <unistd.h>
#include <sys/socket.h>
#include <arpa/inet.h>
#include <netinet/in.h>

#include "nets.h"
#include <libcaer/libcaer.h>
#include "caer_client.h"

#include <assert.h>

#include <signal.h>
// fix stdatomic.h for C++ compat : https://gcc.gnu.org/bugzilla/show_bug.cgi?id=60932
#ifdef __cplusplus
#include <atomic>
using namespace std;
#else
#include <stdatomic.h>
#endif
```

 * caer_client.h

```
#pragma once

#ifdef __cplusplus
extern "C" {
#endif

#include<inttypes.h>
```

 * nets.h (copied from caer repo)

```
/*
 * nets.h
 *
 *  Created on: Jan 19, 2014
 *      Author: chtekk
 */
#ifdef __cplusplus
extern "C" {
#endif

#ifndef NETS_H_
#define NETS_H_

#include <stdlib.h>
#include <stdbool.h>
#include <stdint.h>
#include <unistd.h>
#include <sys/socket.h>
#include <sys/types.h>
#include <fcntl.h>
```

These are all header/includes used in the code...

breznak, 2016-10-17:
... one oddity is that I have to include the `caer_client.c` file to get the caer methods known to C++

breznak, 2016-10-17:
I found this related(?): https://stackoverflow.com/questions/13040139/namespace-issues-in-c11

breznak, 2016-10-17:
```
#define LIBCAER_H_

#ifdef __cplusplus

#include <cstdio>
#include <cstdint>
#include <cinttypes>
#include <cstring>
#include <cerrno>

extern "C" {
#endif

// Common includes, useful for everyone.
#include <stddef.h>
#include <stdlib.h>
#include <stdio.h>
#include <stdbool.h>
#include <stdint.h>
#include <inttypes.h>
//!#include <string.h>
#include <errno.h>

// Use portable endian conversion functions.
#include "portable_endian.h"

// Include libcaer's log headers always.
#include "log.h"

```

@llongi is this the error? there's no #else if branch, so if I have C++, both `c-` and normal vetsions get included (`cstring`, `string`, etc)

breznak, 2016-10-17:
I got a dummy workaround by: 
```
+ #define caerLog(logLevel, subSystem, format, ...)
  #include "caer_client.h"
```

llongi, 2016-10-17:
First, libcaer.h only provides the base infrastructure. It is correct that you then have to include the actual event types you use (events/polarity.h, events/frame.h and so on). Same for the devices if you need those functions.
Also the only place I see you're including anything libcaer related is in caer_client.c. tcpststat.c just does that directly because it's just a small toy utility, usually you'd place those includes inside some header file. In theory the libcaer header files should be C++ compatible already, they do declare the extern C, and do include the needed C++ standard C headers. The conflict with cstring/string.h is very strange, I can compile the examples in examples/dvs128_simple.cpp just fine, with G++ 5.3.1 and Clang++ 3.7.0. Could you try compiling just these and see what happens?

breznak, 2016-10-17:
I can compile both examples just fine in clang/gcc c/c++. 
I still have to apply the "hack" in https://github.com/inilabs/libcaer/issues/8#issuecomment-254233659 to compile in clang++-3.8 (have to remove `cstring`,removing `string` does not resolve it)

llongi, 2016-10-18:
Interesting, so the change in memchr() between the two seems genuine and specified by the standard:
http://stackoverflow.com/questions/8380805/difference-between-string-h-and-cstring
I think introducing an IF/ELSE for the include is what makes the most sense here.

llongi, 2016-10-18:
Fixed.

closed 2016-10-18 by llongi

breznak, 2016-10-18:
Thank you for the fix! I'll test tomorrow if I can remove my hack for `caerLog()`

breznak, 2016-10-22:
I still have the `cstring` x `string` conflict: 

```
In file included from /home/mmm/devel/HTM/nupic.core/src/nupic/utils/network/NetworkReader.cpp:2:
In file included from /home/mmm/devel/HTM/nupic.core/src/nupic/utils/network/NetworkReader.hpp:9:
In file included from /home/mmm/devel/HTM/nupic.core/src/nupic/utils/network/libs/caer/caer_client.h:8:
In file included from /usr/local/include/libcaer/libcaer.h:22:
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/cstring:116:3: error: conflicting types for 'strstr'
  strstr(char* __s1, const char* __s2)
  ^
/usr/include/string.h:337:14: note: previous declaration is here
extern char *strstr (const char *__haystack, const char *__needle)
             ^
```

For `<cstring>` the compiler states the include path (reason, from libcaer), for `<string>` it says nothing..can I figure the include reason for `string`? Unfortunately cstring is required for memcpy, so I cant omit it..

llongi, 2016-10-24:
Hmm one thing I found out is that Clang 3.8 seems to be sensitive to having C++ headers inside the "extern C" declaration, which is why we now only wrap the actual C function declarations and so on with it, but the <cstring> and other includes are outside. Check that you include all C++ headers before, as well as libcaer, outside of any "extern C", and only wrap real pure C headers (if you use any) in it.