inivation/libcaer #10: Move header file inout_common.h from caer to libcaer

breznak, 2017-02-17:
```
../caer/modules/misc/inout_common.h:#define AEDAT3_NETWORK_HEADER_LENGTH 20
```

This file contains many useful definitions, and if I write a 3rd party app, I need to copy&include this file from `caer` to use the constants, moving to `libcaer` is where I believe the definitions belong and would allow to depend purely on libcaer as intended.

llongi, 2017-02-17:
Hi, yes I thought of that too when creating that file, where to place it, but concluded that libcaer would be the wrong place, since libcaer is just concerned with devices and events. It does not define or create any file or network streams or formats, only cAER actually does that, so I kept it close there. I'm open to move it, but not entirely convinced yet.

breznak, 2017-02-17:
Hi Luca, 

> since libcaer is just concerned with devices and events 

Well, I see both sides now... First I thought (HW, events, +format) would make sense for `libcaer`; but then one could continue to add the I/O etc to it. 
On the other hand, if we look at it from the user-programmer view, it would be nice to depend only on libcaer

closed 2017-03-03 by llongi

breznak, 2017-03-03:
Thank you for submitting the patch! 
The header `network.hpp` is not installed during the "make install" phase, is it a mistake?

llongi, 2017-03-03:
Yes, that was a mistake. Commit 43e04ee11b4730cd4435257ff582913a15c02617 fixes this and now network.hpp is installed correctly. Thanks for noticing!

commit 3999422e (2017-03-03): Add network.h include with AEDAT 3.X network stream functions.
Close #10.