inivation/libcaer #14: Bug: libcaer linking error when using C++ headers log

breznak, 2017-04-19:
Env: linux, gcc 4.8, 5.4; clang 3.8, 4.0, 5.0 

I'm including `libcaer(cpp)` headers into my c++ application:
- with C-headers it worked OK (after problems with `extern "C" {}` were ironed out)
- with C++ headers I get a linker error (`libcaercpp/`, I'm including `libcaer.hpp, network.hpp, events/polarity.hpp`) 

Output looks like:
```
XX.cpp.o): In function `libcaer::log::logLevelGet()':
XXX: multiple definition of `libcaer::log::logLevelGet()'
CMakeFiles/XX.cpp:(.text+0x10): first defined here
XX.cpp.o): In function `libcaer::log::logLevelSet(libcaer::log::logLevel)':
XXX.cpp:(.text+0x0): multiple definition of `libcaer::log::logLevelSet(libcaer::log::logLevel)'
CMakeFiles/XX.cpp.o:XX.cpp:(.text+0x0): first defined here
...
```
etc for all `log*()` functions. 
The issue looks slightly similar to former problems in #8 , #12 

I'm using up-to-date libcaer. 

Workaround: 
- I'm able to successfully compile and link my app when I comment out all the "log*" functions in `libcaercpp/libcaer.hpp` (only keeping the include libcaer.c)
- I noticed the C version has a separate file for the logging functionality (`log.h`), if this issue cannot be resolved otherwise, I propose to at least separate this file as well in c++ headers.

llongi, 2017-04-19:
I really can't reproduce this. The log functions being in their own file or not wouldn't change anything, as we'd just include the file unconditionally into libcaer.hpp, as we do for the C version (log.h is always included in libcaer.h). I just tried compiling the examples, and even adding includes for libcaer.hpp and network.hpp manually, still works fine (after updating them to use C++ log()). Could you please paste your include directives that you're using in your code? Also how you link/build. And are you sure you only have one installed libcaercpp in /usr/include, and not /usr/local or similar, and that is being used?

breznak, 2017-04-19:
> I really can't reproduce this.  

That's the problem. I'm not sure how to create a small test-case for this. 
Should this work?
```
File A:
#include <>
class A {
 //empty
}

File B: ...
```
> And are you sure you only have one installed libcaercpp in /usr/include, and not /usr/local or similar, and that is being used? 

Yes, I tried removing it and got an error (so there's only 1)

> Could you please paste your include directives that you're using in your code? 

I'm linking to SWIG, Boost, apr, libcaer (obviously :) ), YAML cpp, numpy, 

> Also how you link/build. 

I build & isntall libcaer, the project builds using cmake - the project used to compile/link OK, even with the libcaer; 

I guess I'll try reverting back to the C(not ++) includes and see which introduces the error (?)

Thank you very much for looking into this!

breznak, 2017-04-19:
@llongi confirmed with c vs cpp includes: 

```
+++ b/MyCaerLib.cpp
@@ -8,8 +8,8 @@
 #include<cassert>

 #include "MyCaerLib.hpp"
-#include <libcaercpp/events/polarity.hpp>
-#include <libcaercpp/events/frame.hpp>
+#include <libcaer/events/polarity.h>
+#include <libcaer/events/frame.h>

 using namespace std;
diff --git a/MyCaerLib.hpp
@@ -1,8 +1,8 @@
 #pragma once

-#include <libcaercpp/libcaer.hpp> //libcaer; use -lcaer
-#include <libcaercpp/network.hpp>
-#include <libcaercpp/events/common.hpp>
+#include <libcaer/libcaer.h> //libcaer; use -lcaer
+#include <libcaer/network.h>
+#include <libcaer/events/common.h>

 #include <vector>
 #include <cinttypes>
```

I just change the includes from `<libcaercpp/*.hpp>` to `<libcaer/*.h>` and it works as expected. 

Does it make sense that I "debug" by mixing cpp/c includes to see where it fails? 
Also, I only changed the includes, is there a need to use some `namespaces` for c++? That could be the catch.

llongi, 2017-04-19:
It should not be, all the C++ files do include the C files too. Also all the include guards and extern C declarations look fine. And there should be no name clashes. The C++ stuff is all properly namespaced in the libcaer namespace, so libcaer::log, libcaer::events::PolarityEvent and so on, while the normal C stuff is namespaced by using the caer prefix on all functions/types, so caerPolarityEvent, caerLog() and so on.

breznak, 2017-04-19:
I have some mixture. 
When I include the C-only, it works OK. 
C++ include fails, but I'm apparently still using the C (exported with the c++) structures, such as
`  struct aedat3_network_header networkHeader = caerParseNetworkHeader(header.data());`

But as you say, all the extern-C, and namespaces should work well, I also doublechecked that.

llongi, 2017-04-19:
I just tried compiling the C example (examples/davis_simple.c) with G++ and changing all the includes from C to C++, but without changing the code, so like:

-#include <libcaer/libcaer.h>
-#include <libcaer/devices/davis.h>
-#include <stdio.h>
-#include <signal.h>
-#include <stdatomic.h>
+#include <libcaercpp/libcaer.hpp>
+#include <libcaercpp/devices/davis.hpp>
+#include <cstdio>
+#include <csignal>
+#include <atomic>
+
+using atomic_bool = std::atomic_bool;
(and memory_order_relaxed to std::memory_order_relaxed).

And it works fine. So even mixing C++ while using the C stuff shows no problems.
It's unlikely there is any conflict with the other C++ libraries you're using, since the C++ support is as I said namespaced in libcaer::*. I don't see what's happening here in your case...

breznak, 2017-04-19:
Thak you, looks like I have to come up with a test case.

llongi, 2017-04-30:
Hmmm I just had and fixed a truly cryptic error because on Windows, the ERROR enum clashed with an ERROR macro defined in some header. Maybe the origin of this problem could be similar, some name clash? Please try latest libcaer from git, and if that still doesn't work, try to add the following to libcaer.hpp right after the includes:

#undef log
#undef logLevel

breznak, 2017-04-30:
Thank you for the heads up, I'll test this!

breznak, 2017-05-17:
I recenty revisited the issue, it's valid for g++5.4, while clang++-3.8 works fine. I am doing LTO optimizations with the g++, so that may be the culpit.

llongi, 2017-05-19:
If you find anything that suggests this is not a GCC+LTO bug (of which a quick search it seems there are many, try GCC 6.3 maybe?), or a way to mitigate the problem with LTO, please reopen.
Thanks and have a good day, Luca.

closed 2017-05-19 by llongi

llongi, 2017-05-30:
I managed to reproduce this and track it down. Turns out in C++ you need the inline keyword for functions that are defined in header files, while you don't need it for class member functions as that's automatic. So the functions for logging and in utils.hpp got indeed defined multiple times.
This is now fixed in release 2.1.4.

breznak, 2017-05-30:
Wow, this is really awesome news, thank you @llongi !! Looking forward to test the new release soon.