inivation/libcaer #18: compilation problem with the current version

Thanh-Binh, 2017-05-11:
I have today problem with the current version of libcaer()
/usr/include/libcaer/network.h:25:30: error: types may not be defined in parameter types
 struct aedat3_network_header {
                              ^
/usr/include/libcaer/network.h:31:3: error: expected constructor, destructor, or type conversion before ‘;’ token
 });
Could you please help me?
 
Here is my header file:
aedat3_network_header caerFormatParser(std::vector<uint8_t> header)
{
  assert(header.size() == AEDAT3_NETWORK_HEADER_LENGTH);
  // Decode network header.
  struct aedat3_network_header networkHeader = caerParseNetworkHeader(header.data());
 return networkHeader;
}

breznak, 2017-05-11:
> struct aedat3_network_header networkHeader = caerParseNetworkHeader(header.data()); 

Is this correct? Shouldn't be just
`aedat3_network_header networkHeader = caerParseNetworkHeader(header.data()); `

Thanh-Binh, 2017-05-11:
You are right! I find out why the word "struct" was inserted into my code by comparing the original and current version of my file. The touchpad problem! Best thank for your best eyes! It is clear here so that i will close this issue!

closed 2017-05-11 by Thanh-Binh