inivation/libcaer #20: [Bug] C++ cast caerEventPacketHeader + vector.data() change

breznak, 2017-05-17:
After work in https://github.com/inilabs/caer/issues/77 , I discovered a sublime bug in libcaer's C++ behavior. 
Or actually it behaves correctly, but it's highly unintuitive. 

Use-case: 
- Working with C++ headers, I read data of packet header (28B), now I need to get `eventType` to construct proper `<typed>EventPacket` (eg `PolarityEventPacket`). 

Code I used:
```
    auto packet = this->net.rcvAll(CAER_EVENT_PACKET_HEADER_SIZE);
    auto header =(caerEventPacketHeader)packet.data();
    // append packet header + data to form a full packet
    const size_t DATA_SIZE = header->eventSize * header->eventCapacity;
    const auto data = net.rcvAll(DATA_SIZE);
    packet.insert(end(packet), begin(data), end(data));
// STEP 3: parse polarity packets
    if ((int)header->eventType == (int)POLARITY_EVENT) { //do your work ...
```

Do you see a bug in there? 
...I wasn't getting any packets processed by my code

```
 // append packet header + data to form a full packet
    const auto data = net.rcvAll(DATA_SIZE);
cout<< "eType="<<header->eventType<<" POLAR_T="<<POLARITY_EVENT<<endl; //1: prints eg eType=1 POLAR_T=1
    packet.insert(end(packet), begin(data), end(data));
cout<< "eType="<<header->eventType<<" POLAR_T="<<POLARITY_EVENT<<endl; //2: always prints eType=0 POLAR_T=1
    header =(caerEventPacketHeader)packet.data(); //hack,fix
cout<< "eType="<<header->eventType<<" POLAR_T="<<POLARITY_EVENT<<endl; //3: correctly prints again eg eType=1 POLAR_T=1
```

I'd like to fix this behavior! Some proposed solutions: 
1. proper C++ `EventPacketHeader` class that owns the resource:
  Expected usage:
```
  auto header = EventPacketHeader(packet);
  cout << header->eventType;
  packet.insert(...);
  cout << header->eventType; //same! 
```
2. allow generic `EventPacket()`
  Which already has a constructor using caerEventPacketHeader + data, but throws assert on "use proper <typed>EventPacket to ensure correct RTTI ..."
  Can we allow to construct with the header only (to access the header values) + add method
  ````
  T castTo<T>(vector data) { 
    assert(T==eventType); 
    vector new_v = headerData_.insert(...data); 
    auto newHeader = (caerEventPacketHeader)new_v.data(); 
     return T(newHeader);
  }
```

breznak, 2017-05-17:
@llongi what do you think? Do you see a better approach?

llongi, 2017-05-17:
The whole issue stems from your usage of a vector to hold the bytes. If you want the raw data, you need to access .data(), and the moment you .insert() on it, it gets reallocated and the old "header" pointer is invalid. There's nothing strange here, you're using a tool and need to be aware of its quirks.
An EventPacket is a block of memory with both a header and its data, and I don't want to break/change that assumption in C++, which is why the various EventPacket constructors expect a "full" event packet, and that will remain an invariant for them. Even if vector were to be supported directly, to avoid having to remember to call .data(), it would expect a full vector with all the bytes. And I'm not convinced it needs to be vector-aware, as that's just one of dozens of representations for "a bunch of bytes".
If you want to avoid having to check the type to build an EventPacket, in libcaercpp/events/utils.hpp there already is a couple of functions doing exactly that: makeSharedFromCStruct() and makeUniqueFromCStruct().
Now I agree that often you only want to look at the header and gets some info from it, so a solution similar to (1) is what I'm thinking about. Of course without the magic vector support.

llongi, 2017-05-17:
After thinking about this some more, I've decided to not go the route of an extra C++ EventPacketHeader. The already present C version and its few, simple functions (caerEventPacketHeaderGet*() and friends) are perfectly sufficient for the job of extracting the few things one might need in this particular case of network clients. I have added caerEventPacketGetDataSize() and caerEventPacketGetSize() to simplify size calculation a bit, so now you can do:

    auto packet = this->net.rcvAll(CAER_EVENT_PACKET_HEADER_SIZE);
    const size_t DATA_SIZE = caerEventPacketGetDataSize((caerEventPacketHeader) packet.data());
    const auto data = net.rcvAll(DATA_SIZE);
    packet.insert(end(packet), begin(data), end(data));
    std::unique_ptr<EventPacket> eventPacket = makeUniqueFromCStruct((caerEventPacketHeader) packet.data(), false);
    if (eventPacket->getEventType() == POLARITY_EVENT) { ... }

That's easier and not so error prone. Also the make*FromCStruct() functions now can forward the 'takeMemoryOwnership' parameters, since memory is owned by your vector in this case.

closed 2017-05-17 by llongi

breznak, 2017-05-17:
Thank you for a very swift fix! And for the example :+1:

breznak, 2017-05-17:
@llongi one more follow up, I using the new code, how would I cast the `EventPacket` to matching `<typed>EventPacket`? 
Say, 
```
std::unique_ptr<EventPacket> eventPacket = makeUniqueFromCStruct((caerEventPacketHeader) packet.data(), false);
if (eventPacket->getEventType() == POLARITY_EVENT) { 
  PolarityEventPacket pp = (PolarityEventPacket)*eventPacket; //this does not work, somehow cast to avoid re-creating PolarityEventPacket
}
```

llongi, 2017-05-17:
The C++11 way to do this is:
    std::unique_ptr<PolarityEventPacket> pp = static_cast<PolarityEventPacket *>(eventPacket.release());
Or if you don't want to keep the unique_ptr but just a basic pointer:
    PolarityEventPacket *pp = static_cast<PolarityEventPacket *>(eventPacket.release());

breznak, 2017-05-17:
Thank you! 
I was just looking at the `makeUniqueFromCStruct()` code, can that be changed to return directly the specific cast? I'm assuming all the `<type>EventPacket`s will have the common `get*()` from `EventPacket` header. 

```
std::unique_ptr<EventPacket> makeUniqueFromCStruct(caerEventPacketHeader packet, bool takeMemoryOwnership = true) {
        switch (caerEventPacketHeaderGetEventType(packet)) {
                case SPECIAL_EVENT:
                        return (std::unique_ptr<SpecialEventPacket>(new SpecialEventPacket(packet, takeMemoryOwnership)));
                        break;
```
So that `auto eventPacket = makeUnique..();` would already be `PolarityEventPacket` (or any other, specific); The user wouldn't have to specify anything, as the type can be recognized from the header.

breznak, 2017-05-17:
> std::unique_ptr<PolarityEventPacket> pp = static_cast<PolarityEventPacket *>(eventPacket.release()); 

That did not seem to work, so I had to use `dynamic_cast`, which I found at 
https://stackoverflow.com/questions/17417848/stdunique-ptr-with-derived-class

```
    if (eventPacket->getEventType() == POLARITY_EVENT) {
auto *tmp = dynamic_cast<PolarityEventPacket*>(eventPacket.get());
std::unique_ptr<PolarityEventPacket> polarityPacket;
if(tmp != nullptr)
{
    eventPacket.release();
    polarityPacket.reset(tmp);
}
```

At the first glance this seems slower (?) than reconstructing the object again as
>     const PolarityEventPacket polarityPacket = PolarityEventPacket((caerPolarityEventPacket) packet.data(), false);

llongi, 2017-05-17:
No, that can't be done automatically. C++ resolves its types at compile time, so you can't change the return type of something based on a runtime integer somewhere in memory, the best you can do is return the closest common type (EventPacket in this case). I'm not sure why the .release() method doesn't work, it's the same we're using for implementing copy(). What is the error?
In the end there should be no slow-down at all for both methods, since the memory is still owned and part of the vector, what you're moving around is just a pointer to it. And sure, you're constructing an EventPacket class, but that is just a pointer and a boolean, VS constructing a std::unique_ptr, which is just a pointer usually. So the differences in memory usage and what's happening are negligible from a performance point of view, though building the std::unique_ptr should be marginally faster (less checks) and uses 4-8 bytes of memory less.

breznak, 2017-05-17:
> I'm not sure why the .release() method doesn't work, it's the same we're using for implementing copy(). What is the error? 

```
error: no viable conversion from 'libcaer::events::PolarityEventPacket *' to
      'std::unique_ptr<PolarityEventPacket>'
std::unique_ptr<PolarityEventPacket> pp = static_cast<PolarityEventPacket *>(eventPacket.release());
                                     ^    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:200:17: note: candidate constructor not viable: no known
      conversion from 'libcaer::events::PolarityEventPacket *' to 'nullptr_t' for 1st argument
      constexpr unique_ptr(nullptr_t) noexcept : unique_ptr() { }
                ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:205:7: note: candidate constructor not viable: no known conversion
      from 'libcaer::events::PolarityEventPacket *' to 'std::unique_ptr<libcaer::events::PolarityEventPacket,
      std::default_delete<libcaer::events::PolarityEventPacket> > &&' for 1st argument
      unique_ptr(unique_ptr&& __u) noexcept
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:356:7: note: candidate constructor not viable: no known conversion
      from 'libcaer::events::PolarityEventPacket *' to 'const std::unique_ptr<libcaer::events::PolarityEventPacket,
      std::default_delete<libcaer::events::PolarityEventPacket> > &' for 1st argument
      unique_ptr(const unique_ptr&) = delete;
      ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:220:2: note: candidate template ignored: could not match
      'unique_ptr<type-parameter-0-0, type-parameter-0-1>' against 'libcaer::events::PolarityEventPacket *'
        unique_ptr(unique_ptr<_Up, _Ep>&& __u) noexcept
        ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/bits/unique_ptr.h:228:2: note: candidate template ignored: could not match
      'auto_ptr<type-parameter-0-0>' against 'libcaer::events::PolarityEventPacket *'
        unique_ptr(auto_ptr<_Up>&& __u) noexcept;
        ^
```

> In the end there should be no slow-down at all for both methods ...

Yes, I observed the slowdown with `dynamic_cast` (which was the only cast I managed), and it's known to be slower than the other casts.