inivation/libcaer #26: libcaer::events::FrameEvent copy is broken

jo-m, 2017-06-13:
Hi

currently, the copy assignment for `libcaer::events::FrameEvent` is broken:
```
// broken
const libcaer::events::FrameEvent copy1 = evt;
// reference works
const libcaer::events::FrameEvent &copy2 = evt;
```
The reason is probably [this definition](https://github.com/inilabs/libcaer/blob/0816e9726d43e7172256b293e7d4748358005e22/include/events/frame.h#L104).

The quick and dirty solution would be to just delete the copy constructor, so programmers cannot use it (and can only use references):

```
FrameEvent(const FrameEvent& other) = delete;
```
However, this is not possible because of [this line](https://github.com/inilabs/libcaer/blob/0816e9726d43e7172256b293e7d4748358005e22/includecpp/events/frame.hpp#L337).

Anyways, the nicer solution would be to have working copy semantics. Any ideas?

llongi, 2017-06-13:
True, this is broken in C++, but any kind of fix (having a deleted or defined assignment operator&constructor) breaks the POD requirement, which is of paramount importance to ensure the memory layout of the C++ class is the same as the underlying C struct where the data actually resides. I can't think of an easy or good solution to this right now.

jo-m, 2017-06-13:
I see the problem. However, we are betraying ourselves and the compiler.
We want the event types to be POD so we have assignment, layout compatibility and memcpy. We make ourselves and the compiler believe the type is in fact POD with `static_assert(std::is_pod<FrameEvent>::value, "FrameEvent is not POD.");` and `uint16_t pixels[1];`. However the fact that `const libcaer::events::FrameEvent copy1 = evt;` is broken shows that assignment is broken (as well as memcpy, except if you manually add up struct+pixels size. However then you can as well use dynamic memory). Possible solutions I see:

* Make the C++ inheriting classes (`FrameEvent`, ...) wrappers containing the struct as a member instead of inheriting from it
* Make the pixels field a const pointer. This would probably be possible because the pixel data will only ever get read, never changed. However the memory management issues would be quite bad.
* Maybe there is some preprocessor trick/template magic which can give an error if anybode tries to use `FrameEvent` the wrong way.
* Write a big fat warning into the docs

best
Jonathan

llongi, 2017-08-14:
1 has significant performance implications. 2 is just a big headache, and pixels are meant to be changed. FrameEvents are also used for users painting things and visualizing them. So I went for 3 and 4: all the frame event documentations (C and C++) notes that you cannot directly copy, and to use instead the new caerGenericEventCopy() function. In C++ also the copy and move operators/constructors are deleted, so if you try to do a = b it fails at compile-time.

closed 2017-08-14 by llongi