inivation/libcaer #27: Unrecognized device (Davis240b) on new machine

breznak, 2017-06-21:
* I have set up a new (ubuntu linux) machine for `caer`, followed the manual, can compile both caer & libcaer installed. 
* running on the new machine gives me an error:
```
2017-06-21 16:03:12 (TZ+0200): CRITICAL: DAVIS ID-1: Failed to open device.
2017-06-21 16:03:13 (TZ+0200): CRITICAL: DAVIS ID-1: Failed to open device.
Total events/second:          0 	 Valid events/second:          0 	 Event counts: Min=           0  Max=           0  Avg=  0.002017-06-21 16:03:14 (TZ+0200): CRITICAL: DAVIS ID-1: Failed to open device.
Total events/second:          0 	 Valid events/second:          0 	 Event counts: Min=           0  Max=           0  Avg=  0.00^C*** Error in `./caer-bin': free(): invalid pointer: 0x00007f5a780022b0 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f5a8d11f7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f5a8d12837a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f5a8d12c53c]
/usr/lib/x86_64-linux-gnu/libuv.so.1(uv_run+0x1e8)[0x7f5a8e2c5fb8]
./caer-bin(caerOutputCommonExit+0x18e)[0x41ee1e]
./caer-bin(caerModuleSMv+0x72)[0x40af32]
./caer-bin(caerOutputNetTCPServer+0xc1)[0x41f4c1]
./caer-bin[0x4252d2]
./caer-bin[0x40972c]
/lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba)[0x7f5a8e9166ba]
/lib/x86_64-linux-gnu/libc.so.6(clone+0x6d)[0x7f5a8d1af3dd]
======= Memory map: ========
00400000-0042d000 r-xp 00000000 08:05 539363                             /home/xxx
0062d000-0062e000 r--p 0002d000 08:05 539363 
```
In dmesg
```
[13806.346675] usb 2-1.2: Product: DAVIS FX2
[13806.346677] usb 2-1.2: Manufacturer: INI
[13806.346680] usb 2-1.2: SerialNumber: 04010098
[13931.963654] usb 2-1.2: USB disconnect, device number 11
[14890.916029] e1000e: enp0s25 NIC Link is Down
[18579.288890] usb 2-1.2: new high-speed USB device number 12 using ehci-pci
[18579.398572] usb 2-1.2: New USB device found, idVendor=152a, idProduct=841b
[18579.398578] usb 2-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[18579.398581] usb 2-1.2: Product: DAVIS FX2
[18579.398584] usb 2-1.2: Manufacturer: INI
[18579.398586] usb 2-1.2: SerialNumber: 04010098
```

llongi, 2017-06-21:
Can your user see the device?
'lsusb -v -d 152a:841b' should give no errors.
Also can you just run the libcaer examples and see if that works? Are you sure only one libcaer is installed?

breznak, 2017-06-22:
> Also can you just run the libcaer examples and see if that works? Are you sure only one libcaer is installed? 

Will do. And yes, it's a completely new system install. 

> Can your user see the device?
'lsusb -v -d 152a:841b' should give no errors. 

I will be able to test this later, but I think it will do the trick! As I forgot to add the `udev` rule: 
In file /etc/udev/rules.d/65-inilabs.rules : 
```
SUBSYSTEM=="usb", ATTR{idVendor}=="152a", ATTR{idProduct}=="84[0-1]?", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="8613", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="04b4", ATTR{idProduct}=="00f3", MODE="0666"
```

Thank you!

llongi, 2017-06-22:
Yes, not having the udev rules would explain this. :-)

closed 2017-06-22 by llongi