inivation/libcaer #29: davis example doesn't work as documented

tobidelbruck, 2017-06-22:
I guess I must be missing something. After sudo make install of libcaer, I cd to examples and run the gcc example, but got error below

hyo@hyo-VirtualBox:~/git/libcaer/examples$ gcc -std=c11 -pedantic -Wall -Wextra -O2 -o davis_simple davis_simple.c -D_DEFAULT_SOURCE=1 -lcaer
davis_simple.c: In function ‘main’:
davis_simple.c:49:52: error: ‘CAER_DEVICE_DAVIS’ undeclared (first use in this function)
  caerDeviceHandle davis_handle = caerDeviceOpen(1, CAER_DEVICE_DAVIS, 0, 0, NUL
                                                    ^
davis_simple.c:49:52: note: each undeclared identifier is reported only once for each function it appears in
davis_simple.c:126:5: error: unknown type name ‘caerPolarityEventConst’
     caerPolarityEventConst firstEvent = caerPolarityEventPacketGetEventConst(po
     ^
davis_simple.c:126:41: warning: implicit declaration of function ‘caerPolarityEventPacketGetEventConst’ [-Wimplicit-function-declaration]
     caerPolarityEventConst firstEvent = caerPolarityEventPacketGetEventConst(po
                                         ^

tobidelbruck, 2017-06-22:
https://github.com/inilabs/caer/issues/76 seems same error from older libcaer install

llongi, 2017-06-22:
Yes this looks like an old libcaer install, where these new features were not yet available.
It's not that difficult to keep things clean, just make sure to always use the same -DCMAKE_INSTALL_PREFIX for libcaer and it will always install in the same place, and overwrite old files as expected. Now some people just don't define that and then you get /usr/local as default, which might fail later since not all distributions do search for libraries and include files there. I always use -DCMAKE_INSTALL_PREFIX=/usr, so that stuff goes into the usual Linux directories, and then can be found as expected. There is no issue here with libcaer, it's just about people maintaining their systems in a coherent, clean way; same as any other self-compiled piece of software. We might at some point offer RPM packages, but that adds non trivial amounts of overhead to the whole process, so it's not a priority.

closed 2017-06-22 by llongi