inivation/libcaer #32: Blocking dataGet() never exits if cable is pulled

jo-m, 2017-07-03:
Hi

Please have a look at the attached modified example: [davis_simple.cpp.zip](https://github.com/inilabs/libcaer/files/1118946/davis_simple.cpp.zip).

* Libcaer version: 2.3.0
* OS: macOS 10.12.5 (16F73)

Running:

    g++ -std=c++11 -pedantic -Wall -Wextra -O2 -o davis_simple davis_simple.cpp -D_DEFAULT_SOURCE=1 -lcaer
    ./davis_simple
    > DAVIS ID-1 SN-00000000 [20:14] --- ID: 1, Master: 1, DVS X: 240, DVS Y: 180, Logic: 9880.
    > New bias values --- PR-coarse: 2, PR-fine: 116, PRSF-coarse: 1, PRSF-fine: 33.
    > before dataGet
    > after dataGet
    > 
    > Got event container with 5 packets (allocated).
    > Packet is empty (not present).
    > Packet of type 1 -> 2 events, 4096 capacity.
    > First polarity event - ts: 28834, x: 85, y: 19, pol: 1.
    > Packet is empty (not present).
    > ...
    *pull USB plug*

Expected output after pulling the plug:

    usbShutdownHandler
    after dataGet
    Shutdown successful.

Output received:

    before dataGet
    *blocking, nothing happens until sending SIGTERM via Ctrl+C*
    ^CglobalShutdownSignalHandler
    after dataGet
    Shutdown successful.

--> usbShutdownHandler is never called.

llongi, 2017-07-03:
Cannot reproduce. On both my home and work Linux machines:
...
Got event container with 5 packets (allocated).
Packet is empty (not present).
Packet of type 1 -> 3277 events, 4096 capacity.
First polarity event - ts: 4488660, x: 75, y: 114, pol: 1.
Packet is empty (not present).
Packet of type 3 -> 1 events, 64 capacity.
Packet is empty (not present).
before dataGet
usbShutdownHandler
after dataGet
Shutdown successful.

jo-m, 2017-07-03:
Hmm, so its probably a macOS issue... will investigate.

jo-m, 2017-07-03:
`davisDataGet()` is caught forever in the `goto retry` loop, because `usbDataTransfersAreRunning(&state->usbState)` never returns `false`. Must be some libusb problem. Investigating further...

llongi, 2017-07-03:
Then it's a MacOS X/libusb problem. The callback should always be called, if the device disappears it should call it with NO_DEVICE. I can confirm this happening on a Mac system here too.

jo-m, 2017-07-03:
[This is my workaround at the moment](https://github.com/jo-m/libcaer/commit/2adc4abe0c48a30ca130baaeaa0ee0e6fa9164d5), I know its ugly..

llongi, 2017-07-03:
If you're sure the callback is never called on unplug (not even with a different error code than NO_DEVICE), please ask on the libusb-dev mailing list and/or open a bug on their github.

jo-m, 2017-07-03:
its called with `LIBUSB_TRANSFER_CANCELLED` repeatedly, but never with `LIBUSB_TRANSFER_NO_DEVICE`.

llongi, 2017-07-03:
So if you take out the '&& transfer->status != LIBUSB_TRANSFER_CANCELLED' checks on both line 567 and 575 then it does call the shutdown handler? If yes, it might be a solution to just do that and then change the doc back to old behavior that the shutdown handler is called on ANY shutdown path, not only the exceptional one.

jo-m, 2017-07-03:
seems to work --> https://github.com/jo-m/libcaer/commit/0f2b7fcef6566c6712c869238d50056c5dea5f25

llongi, 2017-08-09:
Ok great to hear, but working around it that way breaks the USB buffer number/size change support, as it cannot distinguish between an intentional transfer shutdown via libusb_transfer_cancel() and a random shutdown due to any other error, like the device being unplugged.
I've created a test case at https://gist.github.com/llongi/3b50c0559ce71c22aa93b37f24636f0d and would like you to try it on MacOS X. On Linux I get "Status of transfer is: LIBUSB_TRANSFER_ERROR (1)." when unplugging the device, what do you get? Thanks!

jo-m, 2017-08-10:
I ran the test case on current master (c8b7de9).

```
$ gcc -std=c11 -pedantic -Wall -Wextra -O2 -o disconnect_return_test disconnect_return_test.c -D_DEFAULT_SOURCE=1 -lcaer -lusb-1.0.0
$ ./disconnect_return_test
Unable to submit libusb transfer 0. Error: LIBUSB_ERROR_NOT_FOUND (-5).
```

Without a device plugged in, the output is

```
Unable to open device.
```

llongi, 2017-08-10:
Hmmm the test case expects a DAVIS240C to be connected, then you start it and it should just hang there, then you unplug the device while it's running and it should print the transfer exit status. Please come by my office if that doesn't work. :-)

jo-m, 2017-08-10:
With the updated test, the output now is

```
$ ./disconnect_return_test
* hangs until cable is plugged
Status of transfer is: LIBUSB_TRANSFER_ERROR (1).
Status of transfer is: LIBUSB_TRANSFER_CANCELLED (3).
Status of transfer is: LIBUSB_TRANSFER_CANCELLED (3).
Status of transfer is: LIBUSB_TRANSFER_CANCELLED (3).
```

llongi, 2017-08-14:
Can you please test latest libcaer master? I've committed a fix that should work on all libusb versions and supported platforms.

jo-m, 2017-08-14:
Yes, testcase now works as expected on macOS!

llongi, 2017-08-14:
Ok great, and everything's still fine on Linux and Windows too. Closing this.

closed 2017-08-14 by llongi