inivation/libcaer #40: Bias current estimation

jo-m, 2017-09-26:
Hi

In the jAER bias config panel, an estimated ampere value is shown when setting coarse and fine values (I think those estimates are calculated [here](https://github.com/SensorsINI/jaer/blob/60f9b4e7985993d4403479ea75abc3404e61c0b9/src/net/sf/jaer/biasgen/Masterbias.java). Is there any way in libcaer to 1. estimate a current from a coarse/fine value set, and 2. get the best matching coarse/fine value set for a given current?

llongi, 2017-09-26:
Currently this does not exist, but it is in our long-term plans to provide such conversion for Dynapse and DAVIS.

llongi, 2018-09-21:
Added in commit e654c20f5f08619705b231aee42db6387ed57e17.

closed 2018-09-21 by llongi