inivation/libcaer #41: swig compilation error

duguyue100, 2017-09-28:
Hi,

I'm trying to integrate the swig compilation sequence to `pycaer`.
Because I didn't install `libcaer` in system wise, so I append `PKG_CONFIG_PATH` and `pkg-config` can successfully find both `libcaer` and my anaconda python.

When I try to compile using
```
swig -python -I$(pkg-config --variable=includedir libcaer) -cpperraswarn pyflags.i
```

It raises the error:

```
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/libcaer.h:51: Warning 204: CPP #warning, ""Unable to ensure structures are properly packed."".
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/events/common.h:92: Error: Macro 'PACKED_STRUCT' expects 1 argument
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/events/frame.h:85: Error: Macro 'PACKED_STRUCT' expects 1 argument
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/events/frame.h:125: Error: Macro 'PACKED_STRUCT' expects 1 argument
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/events/imu6.h:32: Error: Macro 'PACKED_STRUCT' expects 1 argument
/home/dgywork/workspace/pycaer/libcaer/build/include/libcaer/events/imu9.h:34: Error: Macro 'PACKED_STRUCT' expects 1 argument
```

Do you guys have an idea why this is happening?

llongi, 2017-09-28:
This is a known (to us) bug in SWIG. It chokes on // comments inside a macro.
Use the attached patch on latest github master of SWIG and then it works.
I'll upstream the patch shortly.

[swig.patch.txt](https://github.com/inilabs/libcaer/files/1342066/swig.patch.txt)

duguyue100, 2017-09-28:
Thanks! I will wait for the patch so it's easier to integrate to my submodule.

duguyue100, 2017-09-29:
Thanks Luca and Fede. The problem is solved!

closed 2017-09-29 by duguyue100