inivation/libcaer #45: examples/davis_simple{c,cpp} don't work for a DAVIS640

rochus, 2017-11-09:
Hi,

The example code provided in `davis_simple.c` or `davis_simple.cpp` doesn't work out of the box when using a DAVIS640 (chip ID = 6, Logic: 9912). The call to `caerDeviceDataStart` blocks infinitely. So far I was unable to figure out why this happens. 

As a side note, caer-bin with the minimal davis configuration works without issues and displays both the frames as well as the events. Therefore I suspect some misconfiguration to be the source of the problem.

rochus, 2017-11-09:
The block occurs in dataStart(). It doesn't print anything afterwards and doesn't react to CTRL-C. Changing to non-blocking didn't help neither on master, nor on rpi.

rochus, 2017-11-10:
Ok, the block occurs within dataStart when the DVS is set to run in the call to davisConfigSet at https://github.com/inilabs/libcaer/blob/master/src/davis.c#L2584. 

Changing the order of the calls to davisConfigSet doesn't help either. caer first enables USB (DAVIS_CONFIG_USB_RUN), then MUX and subsequently DVS and APS. Using the same order in libcaer, e.g. moving lines 2596 before 2584, results in the block at the call to davisConfigSet(..., ..._USB_RUN, true);.

Any ideas?

rochus, 2017-11-13:
Looking again into the issue, the differences between the libcaer example and caer start-up is that, first of all, caer is starting in non-blocking mode with all producers disabled. More importantly though, libcaer is missing a setup of the system and usb buffers (functions systemConfigSend and usbConfigSend in caer, see call to functions at https://github.com/inilabs/caer/blob/master/modules/ini/davis.c#L1007). 

Adding these configurations after a caerDeviceSendDefaultConfiguration, the start-up works and I receive packets from the sensor. However, shutdown doesn't operate properly and leaves the device in some dangling state from which it cannot start again. I have to re-plug the sensor. My guess is that the way you use libusb is erroneous and produces a data race somewhere, or (without having access to the the code though) that there's some issue with the state machine on the sensor. If I find some more time I'll try to sort that one out, but I would be glad if you could have a look, too.

llongi, 2017-11-14:
Hi, so the startups are fundamentally equivalent: non-blocking mode only affects dataGet() from its queue, and has no influence on USB. The producers start all disabled but get enabled in the various *ConfigSend() functions, when the "Run" booleans are parsed and sent to the device. libcaer simply starts them all as a convenience for the user.
The systemConfigSend() equivalent is in https://github.com/inilabs/libcaer/blob/master/src/davis.c#L474 and the usbConfigSend() equivalent is in https://github.com/inilabs/libcaer/blob/master/src/davis.c#L526 and https://github.com/inilabs/libcaer/blob/master/src/davis.c#L871.
I cannot look into this deeply until Wednesday, though it all points to the USB async transfer that enables USB on the device never being handled, which is in theory impossible.
So the moment we open a USB device, we start a thread that does nothing but handle all transfers for that device. All configuration functions submit transfers and wait on their completion, and when you do data start a group of transfers for data is submitted and is continuously re-submitted as soon as one's data has been parsed. In your case it looks as if, the moment the data transfers start being handled, the config transfer is just left there and never serviced, hanging the main thread on configSet(). Given the way libusb works and its guarantees, this should never happen, if a transfer is submitted successfully, and there is a thread calling the handling function, at some point it should be handled; the data transfers should never manage to starve or block this.
Does it make any difference if you change libcaer's startProducers order to match cAER? (usb, mux, dvs, aps, imu). It really shouldn't, since by that point the transfers are already there and waiting, in until you enable usb there is no data from the device...
What OS are you running on? What libusb version? Compiler? Anything special about the system (peculiar core/memory configuration, alternative architecture like a SPARC)?

rochus, 2017-11-14:
Ah, thanks for pointing me to lines 474 and 526. I overlooked those two, and it helped me narrow it down further. I (now) think the primary difference between libcaer and caer is that the multiplexer is turned on _before_ the call to davisDataStart. In fact, setting MUX_TIMESTAMP_RUN and MUX_RUN to true before davisDataStart prevents the block (diff here, but I don't think you need it: https://gist.github.com/rochus/3c280cc54c59f3d8248ff52158a436d3). Nevertheless, it leaves the device in some state that a subsequent run of davis_simple immediately jumps into the ubs-shutdown callback (i.e. prints Shutdown successful).

llongi, 2017-11-15:
The 640 I have here works perfectly fine in both libcaer and cAER with no changes. I cannot reproduce this.
The only significant difference I can think of is libusb version, please try an 'emerge =dev-libs/libusb-1.0.21' and see if that changes anything (Hey fellow Gentoo user!).
The order in which you enable USB/MUX/APS/DVS and create the transfers should not matter at all, actually starting the MUX before just means the moment you start USB there already is data to process, which would make any infinite-loop lock-up more, not less likely. The current startup sequence in libcaer first enables USB, at which point due to MUX being off, there is no data that can be sent, so no reason for anything to lock up, the USB_RUN configSet() should return fine, and it does here. The libcaer startup sequence is actually theoretically safer than the cAER one,  since we first make sure everything is on and settled and then enable actual data (MUX_RUN) last.
Can you reproduce this also when connected to another system?

Thanh-Binh, 2017-11-15:
It works perfectly in my Ubuntu 16.04lts. It is difficult to understand with your OS

rochus, 2017-11-17:
The problem occurs with a second davis 640 fx3 as well. Both sensors received a firmware upgrade just last week, so either it's really an issue on my computer, or in the firmware's state machine. Updating libusb to 1.0.21 didn't help. I didn't have time yet to examine the matter on another computer, though. I'll check if there are some unusual configuration settings for the underlying USB subsystem, but I doubt that. For the time being I'll use my workaround (sending MUX_TIMESTAMP_RUN and MUX_RUN before requesting data), so it's not an urgent issue any more. I will let you know if I find the source or an explanation for the problem on the host computer.

llongi, 2018-04-17:
No update in 60 days, closing for lack of information.

closed 2018-04-17 by llongi