inivation/libcaer #48: if Thread support is PThreads: FALSE --> compilation fails

federicohyo, 2017-11-17:
[root@npp libcaer]# cmake -DCMAKE_TOOLCHAIN_FILE=../../toolchain.cmake .
-- Could NOT find Threads (missing:  Threads_FOUND) 
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Project version is: 2.4.0
-- Compiler is Clang: FALSE
-- Compiler is GCC: TRUE
-- Compiler is IntelCC: FALSE
-- Compiler is MS VisualC: FALSE
-- OS is Unix: TRUE
-- OS is Linux: TRUE
-- OS is MacOS X: FALSE
-- OS is Windows: FALSE
-- System is big-endian: 
-- Thread support is PThreads: FALSE
-- Thread support is Win32 Threads: FALSE
-- C flags are:  -std=c11 -pedantic -Wall -Wextra -Wunused -Wundef -Wformat=2 -Winit-self -Wuninitialized -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wredundant-decls -Wmissing-declarations -Wstack-protector -Wshadow -Wfloat-equal -Wconversion -Wstrict-overflow=5 -Wdouble-promotion  -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wjump-misses-init -Wunsuffixed-float-constants
-- CXX flags are:  -std=c++11 -pedantic -Wall -Wextra -Wunused -Wundef -Wformat=2 -Winit-self -Wuninitialized -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wredundant-decls -Wmissing-declarations -Wstack-protector -Wshadow -Wfloat-equal -Wconversion -Wstrict-overflow=5 -Wdouble-promotion 
-- Include directories are:system/include/libusb-1.0;software/libcaer/include/;software/libcaer/includecpp/
-- Library directories are:system/lib
-- Linked libraries are: -lusb-1.0;m
-- Compiled source files are: ringbuffer.c;log.c;events.c;frame_utils.c;usb_utils.c;autoexposure.c;device.c;dvs128.c;davis.c;dynapse.c;davis_rpi.c
-- Final install libdir is: system/lib
-- Final install includedir is:system/include/libcaer
-- Configuring done
-- Generating done
-- Build files have been written to: software/libcaer

when there is no support for p_threads then the compilations fails with... 

software/libcaer/src/usb_utils.c:360:72: error: 'thrd_success' undeclared (first use in this function)
  if ((errno = thrd_create(&state->usbThread, &usbThreadRun, state)) != thrd_success) {

is it possible to compile libcaer without pthreads?

llongi, 2017-11-17:
Threads support is mandatory. On Linux/Unix/OS X this means PThreads. I can't think of a Linux system not supporting that, it looks like the CMake check is simply failing due to cross-compile?.

federicohyo, 2017-11-17:
Yes,  probably.
I'll check on that.

Thank you.

closed 2017-11-20 by llongi