inivation/libcaer #52: Use python to access the device

liwang621, 2017-12-02:
I use libcaer/bindings/python_swig/example/davis.py to read events from DAVIS240C. Because I want to get events histogram, I wrote a piece of code which accumulated events to generate events histogram. But it can not work in real time, and it has huge delays. Can you give me some advice?

llongi, 2017-12-04:
The python code is expected to be less performant than C code, but you should still get close to real-time performance. It really depends on your code, we generally use numpy to handle big arrays of data.
Without seeing your code we cannot help you more, sorry.

closed 2017-12-05 by llongi

liwang621, 2017-12-06:
if __name__ == "__main__":

    import davis
    import cv2
    cv2.namedWindow("DVS",0)
    camera = davis.DAVIS()
    flag = 1
    while flag:
        if camera.read_events() == True :
            img=np.zeros((180,240),dtype=np.uint8)
            for i in range(len(camera.x)):
                if img[camera.y[i],camera.x[i]] < 7:
                    img[camera.y[i],camera.x[i]]+=1
            # print("succesfully read events from the device")
            # print("pixels x:"+str(camera.x))
            # print("pixels y:"+str(camera.y))
            camera.x=[]
            camera.y=[]
            # print("pixels ts:"+str(camera.ts))
            # print("pixels pol:"+str(camera.pol))
            # print("camera frames:"+str(len(camera.frame)))
            cv2.normalize(img,img,255.0,0.,cv2.NORM_MINMAX)
            cv2.imshow("DVS",img)
            cv2.waitKey(1)
#    	else:
#        	print("nothing read from the device")
#     cv2.destroyALLWindows()

The above is my main function, but it can work in real time. Can you give me some advice?