inivation/libcaer #54: Time drift among computer time vs Dynap-se recordings

federicohyo, 2018-01-31:
There is a time difference between the time of a C++ program (via time.h portable time) that sends stimulations and the time returned of the spikes (AEDAT 3.0 recordings). 

Computer time (time.h) says 30s, while recording time says 28.5s.

llongi, 2018-02-02:
While some drift over time is expected, the two timesources run on different crystals, we're talking about 1s over 10 hours (in that range).
Is there any code for me to look at? It's difficult to say where the problem is... libcaer does a very precise time check of the timestamps VS computer time, and warns if the drift goes about 100us, when you compile it with debug mode active (cmake -DCMAKE_BUILD_TYPE=Debug .); first check if that actually complains. If it doesn't, then the problem is in the stimulation code and how it measures time; there's a lot of things one can get wrong there.

llongi, 2018-04-17:
No update in 60 days, closing for lack of information.

closed 2018-04-17 by llongi