inivation/libcaer #55: example daviscvgui.cpp doesn't compile

ru1sb, 2018-02-07:
I'm trying to build daviscvgui.cpp with Qt under ubuntu but I am getting the error:  " ‘const struct libcaer::events::FrameEvent’ has no member named ‘getOpenCVMat’
cv::Mat cvFrame = f.getOpenCVMat(false);" 
I installed libcaer without adding openCv

llongi, 2018-02-07:
Hi, the davis_cvgui.cpp example requires OpenCV support in libcaer to be enabled.
I've added a check now to prevent compilation if it is disabled, and have added a note in the README.
Please recompile libcaer with -DENABLE_OPENCV=1 if you want to use that example specifically.

closed 2018-02-07 by llongi

ru1sb, 2018-02-07:
I did so but I get this errors:
[image]

llongi, 2018-02-07:
Undefined reference means the linker cannot find the appropriate library.
Run 'pkg-config --libs opencv' and make sure the Qt project links to all of those libraries.
I don't use the Qt IDE so I can't tell you where exactly to put that information, but since it links successfully to libcaer (-lcaer), I assume you know where to put library dependencies in Qt IDE. Just do the same as you did for libcaer, but for OpenCV.