inivation/libcaer #58: install prefix to /usr/local in the .cmake file for OS X

liutianlin0121, 2018-02-16:
It seems to me that, for OS X system, one should edit the line `set(CMAKE_INSTALL_PREFIX "/usr")` as `set(CMAKE_INSTALL_PREFIX "/usr/local"`. Otherwise, a "file INSTALL cannot copy file" error occurs even after the sudo option, because OS X won't allow anything not digitally signed into /usr, among other places. See https://derflounder.wordpress.com/2015/09/14/system-integrity-protection-and-the-end-of-xprotect-management-for-browser-plug-ins/.

Also see a similar problem/solution here: https://github.com/openalpr/openalpr/issues/228

llongi, 2018-02-19:
Hi, where are you seeing that line? We do not set the cmake default install path anywhere in the cmake code, and by default it usually is /usr/local.
In the documentation/README we do suggest passing -DCMAKE_INSTALL_PREFIX=/usr to make integration easier (especially pkg-config files in /usr/local doesn't always work reliably), but it's not required and if you omit it it will use /usr/local.
So are you suggesting to change the documentation and hint that on OS X it's preferable to use /usr/local instead?

liutianlin0121, 2018-02-20:
thanks for your reply! The line appears in the generated file cmake_install.cmake. In that file /usr/local was not the used as default in that generated file for some reasons so I changed it manually. Not sure if it is the same situation for OS X users.

closed 2018-02-20 by liutianlin0121