inivation/libcaer #59: Setting Background_Activity_Filter

ru1sb, 2018-03-05:
using this line to set background activity filter: davisHandle.configSet(DAVIS_CONFIG_DVS,DAVIS_CONFIG_DVS_FILTER_BACKGROUND_ACTIVITY,true);
I get following error 
terminate called after throwing an instance of 'std::runtime_error'
  what():  DAVIS ID-1 SN-02460050 [3:14]: failed to set configuration parameter, modAddr=1, paramAddr=29, param=1.
The program has unexpectedly finished.

llongi, 2018-03-05:
Hello, indeed that will fail because DAVIS240C does not support hardware-assisted background-activity filtering (on the CPLD/FPGA).
You first have to check if this is supported by getting a 'struct caer_davis_info' via caerDavisInfoGet() and checking its 'dvsHasBackgroundActivityFilter' member.
I agree this is not well documented, I'll keep this open until I've updated the docs.

ru1sb, 2018-03-05:
so well then what's the best way to get rid of  background noise with DAVIS240C?

tobidelbruck, 2018-03-05:
I'm not sure how to enable the software background filter in cAER, but 
in jAER it must means turning on the default BackgroundActivityFilter 
and setting the correlation time window. See 
https://inilabs.com/support/software/jaer/ , in particular 
https://inilabs.com/support/software/jaer/#h.7hxknxysjx04 , and 
https://sourceforge.net/p/jaer/wiki/Home/ , in particular 
https://sourceforge.net/p/jaer/wiki/Adding%20a%20new%20event%20filter/ . 
These wiki pages are a bit old but they describe how to use the filter. 
This wiki has not been migrated to github yet.

llongi, 2018-03-05:
cAER also has a software background-activity filter (module 'caer_bafilter'), the default config in docs/davis-config.xml enables it (module name 'bafilter').
For pure libcaer uses, we don't currently have a "freestanding" background activity filter. Maybe we'll offer something in that direction in the future.
In the mean-time, just look at the code in https://github.com/inilabs/caer/blob/master/modules/backgroundactivityfilter/backgroundactivityfilter.c#L85 as a pretty concise example of the algorithm. Allocate a 2D map (X x Y sized), put in the timestamps of the events and check if there has been another event in the neighborhood within a certain time window.

llongi, 2018-04-17:
Documentation update is being tracked in issue #43. Closing this one.

closed 2018-04-17 by llongi