inivation/libcaer #61: lib caer api pdf is corrupt

tobidelbruck, 2018-03-18:
https://github.com/inilabs/libcaer/blob/master/docs/libcaer_api_manual.pdf currently points to non-PDF file

llongi, 2018-03-18:
Huh? That is a valid link, it points to the github page and renders the PDF correctly in github. If you press DOWNLOAD it will give you a valid PDF that opens in evince and Acrobat.

tobidelbruck, 2018-03-19:
When I click on the link above, I get a page that starts as follows:  Am I missing something?  When I save the link, I get a  55kB file that is definitely not  a PDF

<!DOCTYPE html>
<html lang="en">
  <head>
    <meta charset="utf-8">
  <link rel="dns-prefetch" href="https://assets-cdn.github.com">
  <link rel="dns-prefetch" href="https://avatars0.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars1.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars2.githubusercontent.com">
  <link rel="dns-prefetch" href="https://avatars3.githubusercontent.com">
  <link rel="dns-prefetch" href="https://github-cloud.s3.amazonaws.com">
  <link rel="dns-prefetch" href="https://user-images.githubusercontent.com/">

  <link crossorigin="anonymous" media="all" rel="stylesheet" href="https://assets-cdn.github.com/assets/frameworks-0b6eee1f89d4460d83bdbee0a4cb0020.css" />
  <link crossorigin="anonymous" media="all" rel="stylesheet" href="https://assets-cdn.github.com/assets/github-c4d5af25254cfc8bb30ae5cb6f074c97.css" />
  
  
  <link crossorigin="anonymous" media="all" rel="stylesheet" href="https://assets-cdn.github.com/assets/site-83dc1f7ebc9c7461fe1eab799b56c4c4.css" />
  

  <meta name="viewport" content="width=device-width">
  
  <title>libcaer/libcaer_api_manual.pdf at master · inilabs/libcaer · GitHub</title>
    <meta name="description" content="GitHub is where people build software. More than 27 million people use GitHub to discover, fork, and contribute to over 80 million projects.">
  <link rel="search" type="application/opensearchdescription+xml" href="/opensearch.xml" title="GitHub">
  <link rel="fluid-icon" href="https://github.com/fluidicon.png" title="GitHub">
  <meta property="fb:app_id" content="1401488693436528">

    
    <meta property="og:image" content="https://avatars3.githubusercontent.com/u/15046222?s=400&amp;v=4" /><meta property="og:site_name" content="GitHub" /><meta property="og:type" content="object" /><meta property="og:title" content="inilabs/libcaer" /><meta property="og:url" content="https://github.com/inilabs/libcaer" /><meta property="og:description" content="libcaer - Minimal C library to access, configure and get/send AER data from sensors or to/from neuromorphic processors. Supported devices are the Dynamic Vision Sensor (DVS), all the DAVIS cameras,..." />

  <link rel="assets" href="https://assets-cdn.github.com/">
  
  <meta name="pjax-timeout" content="1000">
  
  <meta name="request-id" content="AF93:10135:128D20F:2237A37:5AAF72D7" data-pjax-transient>

llongi, 2018-04-17:
That link is to the github PDF, so it gets rendered in GitHub. As I said, click on DOWNLOAD and you'll get the following link which is a normal PDF:
https://github.com/inilabs/libcaer/raw/master/docs/libcaer_api_manual.pdf
The link you mention I cannot find either in our docs or repo, so there's nothing for me to update.
Closing this.