inivation/libcaer #65: Can not import _libcaer_wrap

mmilde, 2018-04-22:
I followed all install instructions to install libcaer and its dependencies. No error during installation occurs. As soon as I try to run the davis.py example (https://github.com/inilabs/libcaer/blob/master/bindings/python_swig/examples/davis.py)

I get the following error
```
In [7]: import _libcaer_wrap as libcaer
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-7-7209eb9fcbb0> in <module>()
----> 1 import _libcaer_wrap as libcaer

ImportError: No module named '_libcaer_wrap'
```
I do can import libcaer using `import libcaer` w/o error. 
I need the python interface to stream the DVS/DAVIS events to a spiking network simulator in python.

Side remark: I tried installing pyaer w/o success, since it depends on libcaer

Cheers and thanks,
Moritz

duguyue100, 2018-04-23:
@mmilde better use my Python bindings, it's much faster and offers more functions than the one available here.
And you don't have to build it yourself.

https://github.com/duguyue100/pyaer

llongi, 2018-04-30:
Indeed the python wrapper is not installed. It is an experimental feature that has to be manually worked with. The _libcaer_wrap.so file has to be where Python can find it, and it's currently your job to ensure that.
To automate this we'd have to integrate it into CMake.
Not sure if worth the effort, especially given @duguyue100 's approach that looks easier for Python users to work with and offers better performance via numpy usage.