inivation/libcaer #70: cmake error on Ubuntu 16.04

CaptainJL, 2018-08-17:
Hello,
I have constantly been getting this cmake error when building libcaer on Ubuntu 16.04.
```

Checking for module 'libserialport>=0.1.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:65 (PKG_CHECK_MODULES)
```

I have no idea how to resolve this.
I have both opencv (via ros kinetic) and libserialport (latest) installed.

Anyone have any solutions/ideas?

Thanks,
CaptainJL

CaptainJL, 2018-08-17:
Just to confirm:
`pkg-config --exists --print-errors libserialport`
Gives no errors

llongi, 2018-08-17:
How did you install libserialport? Ubuntu 16.04 (usually used with ROS Kinetic) does not have the right version of it, only 0.1.0, while 0.1.1 is needed. On such an old Ubuntu either install libserialport manually from source, or don't enable the support for serial devices (-DENABLE_SERIALDEV=0). It is only needed for the eDVS devices.

closed 2018-08-25 by llongi