inivation/libcaer #71: davis.hpp error

raghad1234, 2018-09-05:
Hi,

I cloned the libcaer library on ROS melodic workspace (ubuntu 18.08)  and installed the required packages with the following versions
pkg-config v0.29.1 
cmake v3.10.2
gcc (Ubuntu 7.3.0-16ubuntu3) 7.3.0
libusb-1.0-0:amd64  2:1.0.21-2     amd64          userspace USB programming library

I tried to run the davis_simple.cpp file as the following
g++ -std=c++11 -pedantic -Wall -Wextra -O2 -o davis_simple davis_simple.cpp -D_DEFAULT_SOURCE=1 -lcaer

but I got the following error
davis_simple.cpp:1:10: fatal error: libcaercpp/devices/davis.hpp: No such file or directory
 #include <libcaercpp/devices/davis.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
the davis.hpp file is available but under a different directory

Any clue?

llongi, 2018-09-05:
Hello, did you compile and install the libcaer library itself? The examples show how to use the library, but require that the library is properly installed on your system as detailed in the README.
Does the file /usr/include/libcaercpp/devicse/davis.hpp exist?

raghad1234, 2018-09-05:
I am installing the libcaer library in catkin_ws (since I am using ROS melodic), the davis,hpp file is installed under the following directory
 /home/user/catkin_ws/src/libcaer/includecpp/devices/davis.hpp

llongi, 2018-09-05:
Okay, then you need to pass that special path when compiling, else your system will have no clue where to search outside of a standard location like /usr.
I'd expect passing the following to the g++ call will help:
``-I/home/user/catkin_ws/src/libcaer/``
But then I wouldn't be surprised if it can't find the library because that probably also lives somewhere non-standard.
Did you actually run 'make install', and what path did you pass as -DCMAKE_INSTALL_PREFIX ? That's where things get installed, and if it's not /usr or /usr/local, there's no way GCC can figure it out, so you have to pass the appropriate -I and -L options to it to show where include files and libraries are.

raghad1234, 2018-09-06:
I tried to pass the I/home/user/catkin_ws/src/libcaer/ to the g++ call but the error is still there 
Two questions: 
1- I noticed from the previous installation that the davis.hpp file is under libcaer/includecpp/devices, however, in the path you specified it should be installed under libcaercpp/devices. Is this due to the place of installation? 
2- Is the default path to install the libcaer library and its dependencies  under 
/usr/local/include
? 
Thanks in advance.

llongi, 2018-09-06:
If I remember correctly, the -I option is position dependent, so it would have to come before the source file in the g++ file.
To answer your questions:
1) That is not an installation then. That is where things are inside the git development repo. That is not a valid installation. I don't know what ROS or you are doing there, but it's not the right thing. When you call cmake you need to tell it where to put things (-DCMAKE_INSTALL_PREFIX=/somewhere), if you don't /usr/local is the default usually. Then you call make to compile. And then make install to move things to the aforementioned installation directory, this also shuffles some things around, updates some paths etc., so it's an important step for the library to be truly usable.
2) I recommend /usr as most other tools work best then. For Ubuntu you can also use our repository, please see instructions at https://inivation.com/support/software/libcaer/#ubuntu-linux

closed 2018-09-06 by llongi