inivation/libcaer #72: add cmake packageConfig file with exported cmake configuration

felix2010, 2018-09-20:
As `libcaer` is built with `CMake` it would be helpful if the project would export the cmake-configuration in a package config file `libcaerConfig.cmake`.
This allows `CMake`-based client code to require and use the library simply with
```cmake
find_package(libcaer REQUIRED)
include_directories(${libcaer_INCLUDE_DIRS)
...
target_link_libraries(myapp libcaer::caer)
```

I'm willing to provide a PR, as we already use in our project a patched `libcaer` that provides this feature.

Please let me know if the project's maintainer is interested in this feature.

llongi, 2018-09-21:
Hello, sure, I'd be willing to accept such a PR.
But the way we envisioned and recommend this being done is using the pkg-config tool:

```
INCLUDE(FindPkgConfig)
PKG_CHECK_MODULES(LIBCAER REQUIRED libcaer>=2.4.1)
```

And you get:

```
${LIBCAER_INCLUDE_DIRS}
${LIBCAER_LIBRARY_DIRS}
${LIBCAER_LIBRARIES}
```

Is there a reason this has not been used by you? We already require pkg-config for finding the dependencies, and have had good experiences even when cross-compiling.

llongi, 2018-09-23:
Indeed, but it's a useful additional goodie. I've merged the PR.
I've also raised the minimum CMake version to 2.8.12, as that is what Ubuntu Trusty 14.04 has installed, and that's the oldest system we still support. Thank you for testing with older CMake releases and reporting your findings!

closed 2018-09-23 by llongi