inivation/libcaer #76: `libcaer` can't be found in `/usr/local/lib` when linking

DylanMuir, 2018-10-09:
Hi

I'm having a problem when trying to link against `libcaer`, on OS X (10.13.6). `libcaer` was installed to `/usr/local/lib` with:
```
cmake -DCMAKE_INSTALL_PREFIX=/usr/local .
make
sudo make install
```

I receive an error when linking against the library:
```
ld: library not found for -lcaer
```

Does `libcaer` support installation in `\usr\local\lib`?

Thanks
Dylan Muir

DylanMuir, 2018-10-10:
I am also using `cmake`, `make` to build the final software

llongi, 2018-10-10:
libcaer supports being installed wherever. BUT your system needs to know how to search for it afterwards. Most systems by default only search for libraries in /lib and /usr/lib.
So this is fundamentally something that you have to answer for yourself on your specific system. It's usually enough to pass -I/usr/local/include and -L/usr/local/lib to your compiler and it then happily finds things.
Manual installation is exactly that, and thus leaves these choices, and their consequences, up to the user.
For OS X we highly recommend using the Homebrew repository we offer, which should take care of most of this for you. Please see https://inivation.com/support/software/libcaer/#macos-x for instructions.

closed 2018-10-10 by llongi

DylanMuir, 2018-10-12:
Thanks! I have installed libcaer using homebrew. `cmake` finds the library, but I still need to manually configure `LIBRARY_PATH` and  `CPATH` environment variables to be able to compile and link against `libcaer`. Any suggestions why the files can't be found automatically?